sadekbaroudi/bgkeeb #1: Request for fr4 top plate and backplate

carboncopy101, 2021-12-02:
Hi! Can I ask for a top plate and backplate gerber? This would be very nice to implement since pcb manufacturers make small pcbs for 2USD. Thanks!

sadekbaroudi, 2021-12-02:
[image]

I would have loved to make switch plates, but as you can see, in order to make it fit within the 100x100 limit, I can't fit this switch within the pcb outline. This is fine for the pcb, but if you make a switch plate, I can't fit a whole switch in there. However, I will try to do something that still works. Stay tuned.

sadekbaroudi, 2021-12-02:
Basically, this is what it's going to look like:

[image]

sadekbaroudi, 2021-12-03:
https://github.com/sadekbaroudi/bgkeeb/commit/290e493a76f80711c72a02e8e4b6a4424358d400

sadekbaroudi, 2021-12-03:
go to https://github.com/sadekbaroudi/bgkeeb/tree/master/plates

closed 2021-12-03 by sadekbaroudi

carboncopy101, 2022-01-31:
cool thanks man! Also I had the v1 fabricated. can you link me the steps on how to program the v1 left and right? Thanks again.

sadekbaroudi, 2022-01-31:
> cool thanks man! Also I had the v1 fabricated. can you link me the steps on how to program the v1 left and right? Thanks again.

As it turned out, I was mistaken and the v1 didn't require any special firmware. Build with the current instructions, and it should work fine on the v1 as well. Let me know if you run into any trouble.

carboncopy101, 2022-01-31:
Hello! Nice will update you as soon as I get some free time to work on it.

carboncopy101, 2022-02-01:
Hey! followed your instructions. I got this error:

QMK Firmware 0.14.32
ERROR fingerpunch/bgkeeb: Claims to support a community layout that does not exist: split_3x5_4
Making fingerpunch/bgkeeb with keymap default

anyway it still flashes fine. I tried all the keys except for the encoders. Ill send you pictures when I get and solder the switches. Thanks!

sadekbaroudi, 2022-02-01:
> Hey! followed your instructions. I got this error:
>
> QMK Firmware 0.14.32
> ERROR fingerpunch/bgkeeb: Claims to support a community layout that does not exist: split_3x5_4
> Making fingerpunch/bgkeeb [...]

Interesting. I'll fix that, but glad it still works anyway. Enjoy!