sadekbaroudi/bgkeeb #2: There are not enough holes to place EC11 rotary encoder.

azhizhinov, 2022-02-11:
Hello.
Thanks for your efforts.
Am I right, that 2 legs of EC11 should be removed, bolder ones, because there are no holes for them?
Thanks in advance.

azhizhinov, 2022-02-11:
I'll try. ;-)
[image]
Here the difference beetwin your PCB and the other one with rotary encoder.

sadekbaroudi, 2022-02-11:
Ah, that makes it much clearer! Yes, you can clip those legs. They have no electrical function.

closed 2022-02-11 by sadekbaroudi

azhizhinov, 2022-02-11:
Good.
Thank you.
Maybe a good idea to add that information to your buildguide. ;-)

sadekbaroudi, 2022-02-11:
Fair point! Done: https://github.com/sadekbaroudi/bgkeeb/commit/fb79d040074173b2201d44e4a0736fff486bdede