sadekbaroudi/bgkeeb #5: Can't make them work

robopuff, 2022-03-25:
So I've built one but can't make it work. Both are flashed with left and right settings and when I plug them in separately, they seem to work (both as a left one tho), but whenever I add the TRS cable, it just goes dark - the LEDs turn off and computer doesn't recognize it at all :(

sadekbaroudi, 2022-03-26:
can you share a picture of your build? and how you soldered everything? especially the trrs jacks

robopuff, 2022-03-26:
[image]

[image]

[image]

sadekbaroudi, 2022-03-26:
looks good to me... some more questions...

* What command did you use to build the firmware, and which repository did you use? I can try to flash the same firmware and see what happens on mine
* Are you using a trrs cable, or trs? if trs, it won't work, you need trrs. Also, make sure it's plugged in ALL the way. I had that problem with my adux, and took me two days to figure it out :O

robopuff, 2022-03-28:
Commands used:
```
$ make fingerpunch/bgkeeb:default:avrdude-split-left
$ make fingerpunch/bgkeeb:default:avrdude-split-right
```

but the issue was cable, I've grabbed one from the stack and it had two last pins connected together (dunno why it was made like that, weird premade cable), now that I've used different one, it all works correctly - can share some more photos from finished buld :)

closed 2022-03-28 by robopuff

sadekbaroudi, 2022-03-28:
yay! so glad to hear