sadekbaroudi/bgkeeb #8: LED orientation should honoır the top or bottom?

Ardakilic, 2022-11-12:
Hello,

Which direction's arrows should LEDs honour for orientation?

Which direction should the LED's indicator be same with the tick on the pcb, the back (where we solder), or the front one (where switches will sit)?

Thanks in advance!

Ardakilic, 2022-11-12:
OK, glad it's open source, the led orientation should honor the side that you apply the solder, like the other side has no silk screen. Thanks for making this PCB available for us!

closed 2022-11-12 by Ardakilic

sadekbaroudi, 2022-11-13:
> 

Hi @Ardakilic - My pleasure! Glad you figured it out. Let me know if you have any other questions!

sadekbaroudi, 2022-11-13:
To be clear, the led light should face the opposite direction that you are soldering. This way, the led lighting faces up.

Ardakilic, 2022-11-13:
@sadekbaroudi Heya, first of all thanks! I figured it out, but I was not sure the orientation of the LED, while the orientation is looking up, should honor the top side mark or bottom. That confused me (while you're looking from top side, the mark on the LED and silkscreen do not match)

I also have some issues with encoders with default layout (assuming master is left on default). I soldered the left one horizontally, and the right one vertically. Although the buttons work, the encoders don't. Also the right one works reversed only when right one is connected. I will dig for it further and open an issue if I cannot sort it out.

Thanks again!