sadekbaroudi/bgkeeb #13: wireless mx pls :'(

Merlinov, 2024-03-02:
wireless mx pls :'(

sadekbaroudi, 2024-03-03:
@Merlinov - you can run this wireless as is! nice!nano has inline battery connections, and this keyboard is compatible with the nice!nano footprint. The only real thing missing would be an on/off switch.

Merlinov, 2024-03-03:
I decided that sweep36 with glp ks-33 is the best idea for me. i need some easy to go wireless 36k keeb on which I will not feel discomfort compared to my main one. :'(
[image]

Merlinov, 2024-03-03:
why sweep36 ks-27 didnt have on/off switch to?(

sadekbaroudi, 2024-03-03:
> why sweep36 ks-27 didnt have a on/off switch too?( and what about the cases? are they compatible with ks-27?

sweep36 does have an on/off switch, but it's the swweeep pcb:
https://github.com/sadekbaroudi/sweep36/tree/master/swweeep

That PCB doesn't have KS-27 support though.

> and what about the cases? are they compatible with ks-27?
Yes, choc cases and glp cases are generally compatible with each other. You will need to find MX keycaps with choc spacing though, since this pcb has choc spacing.

Please carefully read through the readme. I have explained all of what you're asking there.

closed 2024-03-03 by sadekbaroudi

atharva-vadeyar, 2024-09-25:
Hi - how do you use the nice!nano to power and control the SK6182 MINI-E LEDs? I thought the LEDs needed at least 3.7V for power and control and the nice!nano can only output 3.3V. I couldn't see a level shifter of any kind on the schematic so not sure how/if this works

Thanks :D

sadekbaroudi, 2024-09-25:
@atharva-vadeyar - You are correct, there is no level shifter. It does technically need at least 3.7v, but I have tested on many boards, and the 3.3v works fine with the mini-e leds :)

atharva-vadeyar, 2024-09-25:
> @atharva-vadeyar - You are correct, there is no level shifter. It does technically need at least 3.7v, but I have tested on many boards, and the 3.3v works fine with the mini-e leds :)
Thank you!

And does the control input for the LED need to come from the Uart Tx or can it come from any GPIO?

atharva-vadeyar, 2024-09-25:
And again, sorry to bug you haha but how does the charging of the LiPo battery work? There isn’t a specific charging socket or a USB port or anything. Where does it get power from?

sadekbaroudi, 2024-09-25:
> And does the control input for the LED need to come from the Uart Tx or can it come from any GPIO?

It can technically come from any, but the pcb is using the uart tx, as you noticed. It's multi-purpose gpio, so it should be fine.

> And again, sorry to bug you haha but how does the charging of the LiPo battery work? There isn’t a specific charging socket or a USB port or anything. Where does it get power from?

The nice!nano has built in charging IC. So you can directly wire the battery to the controller's B+ and B- pins (they are the top ones on both sides, see nice!nano docs). Then when you plug in the nice!nano via usb, it charges the battery

atharva-vadeyar, 2024-09-26:
Hi! Thanks for your help!

I was looking at the pin diagram and I noticed the diagram in the datasheet and the diagram on the footprint you used (and other footprints I've found) for the SK6812MINI-E has the D_IN and D_OUT mixed up. Have you noticed anything similar? Has it affected the LED function in any way? Or have I just found an incorrect datasheet